tronikos/esphome-magnetometer-water-gas-meter #2: Polling rate too slow for most meters

mkmer, 2024-04-23:
Using the 5/8" meter 
@ 30GPM we get 8ms / 1/2 rotation -> much too fast for the polling rate.
@15GPM we get 16ms

Not sure there is much that can be done with this implementation to address the "requirement" to measure 100% of the magnetic pole changes generated by the Neptune meter.

tronikos, 2024-04-23:
Try setting `qmc5883l_update_interval = '5ms'`
That will configure qmc5883l to use 200Hz for its data rate which is the fastest it can go. If that works I can make it the default.
Not sure if you have to modify `qmc5883l_oversampling` too.

tronikos, 2024-04-23:
Setting `qmc5883l_update_interval: '5ms'` still works for me so I can make it the default. Let me know if you also need to reduce `qmc5883l_oversampling`.

tronikos, 2024-04-23:
The 5/8" Neptune T-10 has 0.00864902 (2 / 231.24) gallons per half rotation. At 30GPM that's 30/0.00864902 = 3468.6 half rotations per minute = 57.8 half rotations per second. So one half rotation every 17.3ms. How did you calculate 8ms? Anyway we do need to reduce the update interval to 5ms to accurately sample the sine wave.

mkmer, 2024-04-24:
I believe I had an additional `/2` in there (oops).  
I can also say when we go faster the peak (upper/lower limits) are also much lower - probably related to sampling rate but may also be caused by physics at speed.  The auto tune is setting the values too close to the peaks to work at higher speeds (for me)

elmoret, 2024-04-26:
Oversampling will reduce peak values on fast signals, its an averaging filter after all.

mkmer, 2024-04-26:
I have an in-ground sprinkling system that approaches the maximum volume of the 5/8" meter (I believe it's 35GPM).  One of my zones uses 30GPM - most others use ~25GPM.
I think we are not going to be able to sample the magnet at the max rate first due to the sensor, and second due to the slow overall/variable "scan" rate of ESPHOME.  I'm not sure we can reach the necessary polling rate via the serial bus either.  I was hoping for a magic answer.  This is a very nice solution and I appreciate your efforts.
I may go with AI-on-the-edge type solution.

elmoret, 2024-04-27:
Good to see the appreciation for tronikos work!

Maybe this would help?

https://esphome.io/api/classesphome_1_1_high_frequency_loop_requester#:~:text=Usually%20the%20ESPHome%20main%20loop,loop%20run%20continuously%20without%20waiting.

I assume you’re running on ESP32?

tronikos, 2024-04-27:
I sent https://github.com/esphome/esphome/pull/6647 for review that will make fast update intervals work.

Until that is merged you can try the following config:

```yaml
substitutions:
  qmc5883l_update_interval: '5ms'
  qmc5883l_oversampling: '64x'

esphome:
  name: water-meter
  friendly_name: Water meter
  on_boot:
    then:
      - lambda: App.set_loop_interval(0);
```

I tried this on ESP8266 and ESP32 by also adding:

```yaml
debug:
  update_interval: 5s

sensor:
  - platform: debug
    loop_time:
      name: "Loop Time"
```

The loop time on ESP8266 was ~13ms and on ESP32 ~5ms.
Removing the `api` component the loop time on ESP32 was ~1ms. Not sure if MQTT would be faster. Worth trying.

tronikos, 2024-04-28:
I included the above changes in the repo. See https://github.com/tronikos/esphome-magnetometer-water-gas-meter/commit/e91add14e6ceb76ba344541264c7475db4bb4b36

Please report what's the maximum GPM flow rate you can now measure to mention it in the README.

closed 2024-04-28 by tronikos

tronikos, 2024-04-29:
I added the following code in `QMC5883LComponent::update`:

```cpp
  static uint32_t prev_millis = millis();
  static uint32_t i = 0;
  if (i % 100 == 1) {
    ESP_LOGI("QMC5883LComponent::update", "actual update interval in millis %d", millis() - prev_millis);
  }
  i++;
  prev_millis = millis();
```

The actual update interval of qmc5883l is: 6ms on ESP32, 10ms on ESP8266 with the default loop interval of 16ms.
With a loop interval of 0ms I got: 5ms on ESP32, 8ms on ESP8266.

So if you are using an ESP32 you should be able to get the fastest refresh rate of the sensor.

mkmer, 2024-04-29:
Is the QMC chip in continuous sample mode or single sample mode?  (Is the ESP32 triggering measurements, or just polling the values generated from the chip.)  If it's in continuous mode, what rate is it at (range appears to be 10hz(100ms) - 200hz(5ms) based on spec sheet)?
The issue I have been seeing is the peak flux values (top and bottom) become smaller as the meter speeds up.  The "automatic" learn mode is not subtracting enough AND the middle hysteresis becomes smaller. 
https://github.com/tronikos/esphome-magnetometer-water-gas-meter/blob/9b1dd8095e68c91f371ef95e6699dcdad4c9f070/esphome-magnetometer.yaml#L238

There is a practical limit to the size of the hysteresis, when it's too small and the meter stops in "just the right spot" it results in continuous counting.

tronikos, 2024-04-30:
It's in continuous mode with output data rate at 200Hz.

Make sure you use the latest changes in the repo. See the readme how to use refresh set to 0s for the package to make sure you always pull the latest changes. Make sure you place the sensor well in a spot you get good readings with at least one axis having a wide range. Try on ESP32 and with a light slow stream of water when you calibrate. Then report back what thresholds the calibration computed and what's the maximum flow rate you can measure.

mkmer, 2024-05-03:
OK - update is in.  I had a bit of trouble with the auto calibration, for some reason we get `0` values in the samples, probably related to timing of the register reads in continuous mode when the device is updating said registers?  Maybe 0 is really not "valid".  It's a bit strange because I don't seem to have spurious 1/2 rotation counts in "normal" operation.  Not exactly sure where the 0 came from in the calibration.

Either way, once I manually adjusted for the "invalid" range of -117 to 0 (it really should have been -117 to -40) it appears to be catching ~30GPM. 

I also had to "center" the hysteresis with only a small range (upper/lowers +/- 4 around '-77' - I believe we can only tell frequency NOT amplitude when polling so close to the Nyquist limit requiring a tighter range...  Something larger than +/-4 may work,  I didn't research more once it was working reliably.

tronikos, 2024-05-04:
The 0 value is a bug only during calibration.
In C++ `std::max(std::numeric_limits<float>::min(), -40.0f)` returns 1.17549e-38, almost 0, when it should had been -40.0.
I'll fix the code by using -9999 instead.

tronikos, 2024-05-04:
Great to hear you got it working!
In the latest version I fixed the 0 values, see previous comment, and I changed the code to pick thresholds +-5 around the midpoint. It makes sense for you to need a more narrow threshold range for high flow. The period of your sine wave at the highest flow is 17.3ms and we are sampling every 5ms. It's unlikely the samples to be close to the highest or lowest point.
As long as the threshold range is larger than the noise amplitude we are good. A range of 10 should be good.