tronikos/esphome-magnetometer-water-gas-meter #4: Axis and threshold values are reset every time the ESP32 is disconnected

sesardelaisla, 2024-04-30:
Hello,

I have realized that threshold values are reset every time that my ESP32 ATOM Lite is disconnected or restarted for any reason. Once the ESP32 is back, values are not persistent. For the moment, I have created an automation for the purpose of setting the right values for axis, low and right thresholds every time such values change. However, I consider this as a workaround. Would it be possible for these values to be consistent?

TIA

tronikos, 2024-05-01:
Fixed in https://github.com/tronikos/esphome-magnetometer-water-gas-meter/commit/0212843c97f33d454f3a4fa923510077e8baafe1

closed 2024-05-01 by tronikos

jbronikowski, 2024-07-24:
This issue seems to happen on the latest build. Just lost power to my esp and this caused all settings to go back to default

micle546, 2025-05-11:
My axis is also resetting on power cycle with the latest build

tronikos, 2025-05-12:
Do you have the following per the README?

```yaml
preferences:
  flash_write_interval: 60min
```