tronikos/esphome-magnetometer-water-gas-meter #6: Half rotations total not increasing unless being calibrated

jac2424, 2024-05-03:
Thank you for releasing this project. Have been trying to get it to capture my Neptune water meter readings and eventually my gas meter as well.

Running into an issue where the Half rotations total only increases while calibration is being run. The water flow is also detected at this time. Once calibration completes, water flow shows 0 and Half rotations total gets stuck at what ever it increments to while calibrating and total gallon also gets stuck at where it read during the calibration.

Enabling the data for each axis shows it’s changing when water is running but the other values just don’t update unless calibration is running. Any ideas what could be the issue?

tronikos, 2024-05-04:
This is strange. During calibration the half rotations total and flow should NOT be increasing.

Make sure you calibrate while you are running water at a low flow rate. Can you try with the latest version (as long as you have `refresh: 0s` you will get the latest version after clicking install)? There was a bug with 0 values that I just fixed.

In the logs you should see something like:
```
[22:54:34][I][calibration:203]: x ranged from 13.39 to 76.49
[22:54:34][I][calibration:209]: y ranged from -81.57 to -3.85
[22:54:34][I][calibration:215]: z ranged from -18.03 to 74.43
[22:54:34][I][calibration:238]: Selected z axis with lower threshold: 41.23 and upper threshold: 51.23
[22:54:34][D][select:062]: 'Axis' - Setting
[22:54:34][D][select:115]: 'Axis' - Set selected option to: z
[22:54:34][D][select:015]: 'Axis': Sending state z (index 2)
[22:54:34][D][number:054]: 'Threshold lower' - Setting number value
[22:54:34][D][number:113]:   New number value: 41.200001
[22:54:34][D][number:012]: 'Threshold lower': Sending state 41.200001
[22:54:34][D][number:054]: 'Threshold upper' - Setting number value
[22:54:34][D][number:113]:   New number value: 51.200001
```

And the corresponding entities in HA should get updated to reflect these numbers.

tronikos, 2024-05-04:
Can you post a screenshot like the first one I have in the README before and after clicking on calibrate? Can you also post logs when calibration happens like the logs I posted above?

jac2424, 2024-05-04:
Thanks for the help, will post the logs data later today soon as I can. For now I seem to have it reading the water flow after reinstalling with latest version like you mentioned. One odd behaviour noticed during auto calibration was it chose Z axis but range was more appropriate for Y. Z axis would not show any flow data but manually selecting Y seemed to work. Will still need to verify accuracy but attest when I have water running it detects and updates the number now.

jac2424, 2024-05-04:
Please see attached screenshots and log requested. 

[image]
[image]
[image]

[logs_water-meter-reading_logs.txt](https://github.com/tronikos/esphome-magnetometer-water-gas-meter/files/15210506/logs_water-meter-reading_logs.txt)

tronikos, 2024-05-04:
I see:

```
[13:44:00][I][calibration:204]: x ranged from -40.07 to -27.96
[13:44:00][I][calibration:210]: y ranged from 11.85 to 33.21
[13:44:00][I][calibration:216]: z ranged from -115.29 to -68.49
[13:44:00][I][calibration:239]: Selected z axis with lower threshold: 18.40 and upper threshold: 28.40
[13:44:00][D][select:062]: 'Axis' - Setting
[13:44:00][D][select:115]: 'Axis' - Set selected option to: z
[13:44:00][D][select:015]: 'Axis': Sending state z (index 2)
[13:44:00][D][number:054]: 'Threshold lower' - Setting number value
[13:44:00][D][number:113]:   New number value: 18.400000
[13:44:00][D][number:012]: 'Threshold lower': Sending state 18.400000
[13:44:00][D][number:054]: 'Threshold upper' - Setting number value
[13:44:00][D][number:113]:   New number value: 28.400000
[13:44:00][D][number:012]: 'Threshold upper': Sending state 28.400000
```

The z axis is better than y. The thresholds are wrong though. I fixed this yesterday so it seems you built during the short time I introduced this bug. Please try again with the latest version.

closed 2024-05-04 by tronikos

jac2424, 2024-05-04:
Installed the latest version and the Z axis being auto selected after calibration seems to be working now. Will monitor next few days to see how accurate the readings are when compared to the data from my utility provider, thank again for your help and for this great tool.

tronikos, 2024-05-08:
I forgot to mention. If you want, you can use the same ESP device for both meters. You need to include `esphome-two-meters.yaml` instead of `esphome-water-meter.yaml`. That config is untested so let me know if it works or not.