tronikos/esphome-magnetometer-water-gas-meter #21: Esp8266 Slow Response

jbronikowski, 2024-07-14:
I noticed on the NodeMCU 8266 that I could not achieve over 3.5gal per minute. I had to set logging to info in order for the device to properly read. Have you noticed this before in your setup? By default it debugs everything.

jbronikowski, 2024-07-14:
Closing due to discussions have this already

closed 2024-07-14 by jbronikowski

tronikos, 2024-07-14:
My understanding is logging slows down the device only if you are connected and viewing the logs. Is that not the case? If not, let me know to set the default value to info.

tronikos, 2024-07-14:
Please also try the suggestions in https://github.com/tronikos/esphome-magnetometer-water-gas-meter/discussions/19 that it seems you already discovered. From that discussion I'm planning on making the magnetic fields sensors and rotations count sensor internal by default. Let me know what improvements you see. Let me know what's the maximum flow rate you can measure with the ESP8266 to update the README.

jbronikowski, 2024-07-14:
The flow at the time of measurement was 3.8 with no logging setting and without looking at the log viewer. This was observed within HA. After setting it to INFO I was able to get achieve 15.8. I will be doing some more tests this coming week to see maximum but that is roughly what my meter can produce from our community well.