tronikos/esphome-magnetometer-water-gas-meter #22: ESP32 fails to boot with latest esphome-gas-meter.yaml update on July 14

jac2424, 2024-07-14:
Thank you for your continued work on this. Seems something broke with your update today for esphome-gas-meter.yaml. After the update ESP32 seems to hang at boot. It gets far as resetting boot loop counter and then stops responding. When I revert it back, it works fine again. Not sure if logs below will be helpful but if there is anything else I can provide to troubleshoot, let me know.

```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/gas-meter-reading.yaml...
INFO Updating https://github.com/tronikos/esphome-magnetometer-water-gas-meter@main
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing gas-meter-reading (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- noise-c @ 0.1.4
RAM:   [=         ]  12.6% (used 41304 bytes from 327680 bytes)
Flash: [=====     ]  52.4% (used 961849 bytes from 1835008 bytes)
========================= [SUCCESS] Took 4.26 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.30.9
INFO Uploading /data/build/gas-meter-reading/.pioenvs/gas-meter-reading/firmware.bin (967616 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.14 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.30.9 using esphome API
INFO Successfully connected to gas-meter-reading @ 192.168.30.9 in 11.362s
INFO Successful handshake with gas-meter-reading @ 192.168.30.9 in 0.117s
[13:50:02][I][app:100]: ESPHome version 2024.6.6 compiled on Jul 14 2024, 13:21:46
[13:50:50][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
```

tronikos, 2024-07-14:
The latest update set the log level to info to speed up the device. Not seeing anything in the logs is normal. Are the flow and total sensors not available?

jac2424, 2024-07-14:
Confirmed its working, I thought ESP32 was hanging as after flash ESPHome showed no logging info, didn't realize that would get disabled by the update. Sorry about the false alarm.

closed 2024-07-14 by tronikos