tronikos/esphome-magnetometer-water-gas-meter #31: Possible to use a single ESP 32

spydazcomputers, 2024-09-13:
Relatively new to esphome platform. Would it be possible to pull two sensors from the same processor?

tronikos, 2024-09-13:
https://github.com/tronikos/esphome-magnetometer-water-gas-meter/discussions/28

closed 2024-09-13 by tronikos

spydazcomputers, 2024-09-20:
I have tried to get this to work, understanding that it is not tested however I have failed.

I can get the two meters to read individually using the gas and water yaml's however when I switch to the below code I get an error in the log buffer of i2c failed q*** failed.

Any thoughts, For now I am just using 2 separate esp32 devices but the meters are 3 feet apart using one would be ideal.

substitutions:
  i2c1_sda: GPIO6
  i2c1_scl: GPIO5
  i2c2_sda: GPIO22
  i2c2_scl: GPIO23

packages:
  meter:
    url: https://github.com/tronikos/esphome-magnetometer-water-gas-meter
    ref: main
    file: esphome-two-meters.yaml
   refresh: 0s