tronikos/esphome-magnetometer-water-gas-meter #34: Two Meters not working

spydazcomputers, 2024-09-20:
Based on the post tagged in #31  referring to #28, I have tried to get it working but get an i2c failure.

with both meters connected I can flash just water or just gas and it works fine. 

below is my two-meters code

I have tried to get this to work, understanding that it is not tested however I have failed.

I can get the two meters to read individually using the gas and water yaml's however when I switch to the below code I get an error in the log buffer of i2c failed q*** failed.

Any thoughts, For now I am just using 2 separate esp32 devices but the meters are 3 feet apart using one would be ideal.

substitutions:
i2c1_sda: GPIO6
i2c1_scl: GPIO5
i2c2_sda: GPIO22
i2c2_scl: GPIO23

packages:
meter:
url: https://github.com/tronikos/esphome-magnetometer-water-gas-meter
ref: main
file: esphome-two-meters.yaml
refresh: 0s

tronikos, 2024-09-20:
So  you have connected one magnetometer to the 5 and 6 pins and another to 22 and 23 pins and it doesn't work? What's the exact error that you get? And without any rewiring, just changing the firmware to use one of the 2 magnetometers it works?

spydazcomputers, 2024-09-21:
> So you have connected one magnetometer to the 5 and 6 pins and another to 22 and 23 pins and it doesn't work? What's the exact error that you get? And without any rewiring, just changing the firmware to use one of the 2 magnetometers it works?

Yes, I have 3 firmwares, water only, gas only and water+ gas. Individual ones work the combined does not.

I will reconnect and reflash to get the exact error and post it here

jdandrea, 2024-12-28:
I am wanting to use a single ESP32 as well. I will see if I can replicate once I get all the parts. Or if I get it working I’ll share that here too.

dnschneid, 2025-01-14:
@spydazcomputers are you still having this issue?  If so, please post logs along with as much of your three configs as you are willing to share.

tronikos, 2025-01-15:
Looking at the working https://github.com/dnschneid/espleak/blob/main/espleak.yaml which also uses 2 qmc5883l, the current configuration in esphome-two-meters.yaml is very similar in regards to setting up 2 i2c and 2 qmc5883l. The only difference is:

```
# Pull in external versions of i2c and qmc5883l with added resiliency features
external_components:
  - source: github://dnschneid/esphome@i2creset
    components: [ i2c, qmc5883l ]
    refresh: 60s
```
Maybe some changes there are required?

@spydazcomputers it's worth adding these lines in your config to test it out.

@dnschneid without your changes in these components, can you still connect to 2 qmc5883l? Do you think you could try out esphome-two-meters.yaml just to see if you can read data from both sensors?

dnschneid, 2025-01-16:
Those changes are not required for two qmc5883l, and in fact for most of my prototyping I was just using the standard version.  They just add some features that allow the sensors to be power cycled.

tronikos, 2025-01-16:
Then I guess what pins you pick for the 2 i2c matters? Do you think you could try the esphome-two-meters.yaml with your current pins? If it works with your pins I could default to those and add a comment that other pins might not work.

dnschneid, 2025-01-16:
The ESP32's I2C driver is flexible, so pin selection is only a concern if they are strapping pins, or are missing pull-ups (or have pull-downs) on the board, or are also hooked up to something else.  5/6/22/23 are not strapping pins, so it is really a board-specific thing.  [The board I use](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/resources/ESP32-POE-ISO-GPIO.png) for instance doesn't even have GPIO6 pinned out (it's used for the on-board flash chip).

OP hasn't responded in 4 months so I don't think it's worth the time to try random things to reproduce a problem that might just be a wiring or configuration issue.  With the details we have, for all we know they could have used only the i2c1 pins in their individual testing, and the second sensor isn't wired up correctly at all.  Happy to be proven wrong, of course.

closed 2025-01-19 by tronikos