tronikos/esphome-magnetometer-water-gas-meter #35: Gas Meter Reporting in Gal, instead of ft3 in home assistant

jsb2092, 2024-09-27:
My YAML has a substitution for the units to be ft3:
substitutions:
  volume_unit: 'ft³'
  unit_of_measurement: 'ft³'
  i2c_scl: GPIO4  # D1
  i2c_sda: GPIO3  # D2
  # Set to false only if needed during manual calibration.
  # Do not keep them at false since these slow down the ESP device
  # and reduce the accuracy during high flow.
  hide_magnetic_field_strength_sensors: 'true'
  hide_half_rotations_total_sensor: 'true'

packages:
  meter:
    url: https://github.com/tronikos/esphome-magnetometer-water-gas-meter
    ref: main
    file: esphome-gas-meter.yaml
    # Or for gas meter:
    # file: esphome-gas-meter.yaml
    # Or if you are using HMC5883L instead of QMC5883L:
    # files: [esphome-water-meter.yaml, hmc5883l.yaml]
    refresh: 0s
    

However, in Home Assistant, it is reporting gal for the unit.

[image]

jsb2092, 2024-09-29:
This appears to be a HA issue, closing

closed 2024-09-29 by jsb2092

tronikos, 2024-09-29:
You can manually change the unit in HA in the settings of the entity.