tronikos/esphome-magnetometer-water-gas-meter #39: Y axis values.

atoulmin, 2024-11-11:
Hi,  I’m wondering if someone can help me. I’m struggling to get my sensor dialled in. The auto calibrate doesn’t work for me.  I’m using the HMC5883L sensor. 

I’ve tried running a small amount of gas and look at the numbers. I have the low set to 25 and the upper set to 35.  

This is what the Y axis reporting. 

[image]

are my settings right?

nosoftwareskills, 2024-11-13:
It looks like you have the same issue as what I am seeing on my water meter. The sensor magnetic field is drifting. 

FYI, I think the "issues" section is for software bugs (issues with the posted software). I don't think this is a software bug. It is likely a problem with our setup or a problem with the sensors we are using. If that's the case, a more appropriate place for this may be a discussion. I created a discussion for my issue. Feel free to add any comments there. 
https://github.com/tronikos/esphome-magnetometer-water-gas-meter/discussions/40#discussion-7476748

tronikos, 2024-11-18:
Closing since this is not issue of the code.

closed 2024-11-18 by tronikos