tronikos/esphome-magnetometer-water-gas-meter #58: Few Issues

steve3j, 2025-03-17:
1. Calibrate Axis button in device info doesn't seem to do anything except pop a message up.
"Gas Meter ESP8266 Calibrate axis changed to March 16, 2025 at 6:36 PM triggered by action Button: Press"

2. After power cycling the esp8266, home assistant no longer has the axis, threshold upper/lower, or temp offset saved.

3. The temp offset is in degC when actual temp is in degF. No option on the offset to change it to degF. 

4. Does the uT reading get adjusted based on the temp? I believe uT would drift depending on temp unless it's compensated based on temp.

steve3j, 2025-03-17:
I might be dumb, just saw the discussion page. Also just saw that I need to setup reading/writing to/from the 8266's memory.

tronikos, 2025-03-24:
3. Internally ESPHome uses Celsius and Home Assistant automatically converts it to whatever unit you have in your settings based on your location.
4. Per the datasheet: "Temperature compensation of the measured magnetic data is enabled by default at the factory. Temperature measured by the built-in temperature sensor will be used to compensate the magnetic sensor’s sensitivity changes due to temperatures. The compensated magnetic sensor data is placed in the Output Data Registers automatically."

> How do I get sensor total to show as cubic Ft? Even with Volume_units set to cubic Ft, the total shows as gal.
Home Assistant is likely converting the unit based on your location. Change the unit of measurement of the sensor in Home Assistant.

closed 2025-03-24 by tronikos

Anathema-Device, 2025-03-24:
FWIW, I have rather the same issue.  I have everything that I can find set to cubic M for gas but the gasmeter total value is seen by HA as L so when it is added to the Energy tab HA complains about it not being in an acceptable unit.  But I can't find anything that references L!  My guess is that it is something residual from the watermeter side of things.

tronikos, 2025-03-24:
Try removing the device in Home Assistant and adding it again.